probatio/probatio-3dprinting #10: Parametrize 3D model to adapt to differently calibrated 3D printers

jtragtenberg, 2020-12-04:


DocSunset, 2021-03-25:
This may remove the need for #12 for instance.

DocSunset, 2021-03-25:
Is the idea of "parametrize" that a given user making blocks will tweak parameters of the model to better work with their printer? Or is the goal that the model is flexible and redundant enough that tweaking isn't needed and the model works well across printers?

jtragtenberg, 2021-03-25:
The idea of this issue was to tweak parameters of the model to better work with their printer. But if we found a way to have a model that fits with any printer it would be better

DocSunset, 2021-03-25:
What sort of parameters would be tweaked?

jtragtenberg, 2021-03-25:
Block dimensions (to only change a few milimeters decimals), but this could be asily changed using the f3d file