probatio/probatio-3dprinting #11: Improve Blocks 3D models for better adhesion

jtragtenberg, 2020-12-04:


DocSunset, 2021-03-25:
Any ideas how to make this improvement? Bigger magnets? Mating surface features? Snap fittings between blocks? What did you have in mind?

jtragtenberg, 2021-03-25:
I am testing now thicker magnets. I will see if it is still easy to remove a block with them. @filipecalegario has some cool ideas with mechanic supports.

DocSunset, 2021-03-25:
> I am testing now thicker magnets. I will see if it is still easy to remove a block with them. 

An alternative to thicker magnets is to use multiple thin magnets.
- I think the change in attraction is very similar to using a single magnet as long as the thickness of the stack of magnets is similar
- you can glue just one magnet and later reclaim the rest stacked on top, easier for assembly and maintenance (probatio/probatio#8)
- You can adjust the strength by adding and removing magnets
- It may also be easier to source thinner magnets?

> @filipecalegario has some cool ideas with mechanic supports.

Interested to hear more about mechanic supports to improve adhesion. Are these separate from possible magnet supports for semi-permanent fixing? (#15)

jtragtenberg, 2021-03-25:
Yes, stacking thinner magnets would work. But buying a thicker one is cheaper than buying many thin ones. We can let the maker choose between this cost/flexibility tradeoff

DocSunset, 2021-03-25:
Good idea.