probatio/probatio-3dprinting #12: Define a 3D printer calibration protocol to improve the Block snap-fittings

jtragtenberg, 2020-12-04:


DocSunset, 2021-03-25:
It may also be possible to design the snap fittings so they are less sensitive to poor print quality.

jtragtenberg, 2021-03-25:
My printer was systematically underextrusion and the layers were always systematically thinner. I tried everything to solve that and I could only fix this problem by buying a new extruder/hot end system now that I got money from the project.

I can’t think of a way for a model design that would work equally for a printer with such a problem without redesigning the model. Do you? That would be perfect.

DocSunset, 2021-03-25:
No, you’re absolutely right. We have to be able to assume that the printer is “working properly” at some baseline level. So is the idea of this issue to document a reasonable way of checking that your printer’s basic functionality is all working?